victronenergy/node-red-contrib-victron #238: [BUG] Logging gets flooded

keeskode, 2025-06-02:
**Describe the bug**
When running the node-red plugin newest version ()

It's flooding my deamon.log in my system (raspberry pi).

Before the newer version it wasnt flooding. It now eats up entire diskspace quickly

example log:

Jun 02 10:59:04 2025-06-02T08:59:04.579Z node-red-contrib-victron:dbus ItemsChanged { serial: 59653, path: '/', interface: 'com.victronenergy.BusItem', member: 'ItemsChanged', signature: 'a{sa{sv}}', sender: ':1.10977', type: 4, flags: 1, body: [ [ [Array], [Array] ] ] }
Jun 02 10:59:04 2025-06-02T08:59:04.579Z node-red-contrib-victron:victron-client [MESSAGE HANDLER] com.victronenergy.dcdc/278:/Dc/0/Voltage | { "changed": true, "path": "/Dc/0/Voltage", "value": 13.350000381469727, "text": "13.35V", "senderName": "com.victronenergy.dcdc", "deviceInstance": 278 }
Jun 02 10:59:04 2025-06-02T08:59:04.580Z node-red-contrib-victron:victron-client [MESSAGE HANDLER] com.victronenergy.dcdc/278:/Dc/In/P | { "changed": true, "path": "/Dc/In/P", "value": 5.349999904632568, "text": "5W", "senderName": "com.victronenergy.dcdc", "deviceInstance": 278 }
Jun 02 10:59:04 2025-06-02T08:59:04.614Z node-red-contrib-victron:dbus ItemsChanged { serial: 3435144, path: '/', interface: 'com.victronenergy.BusItem', member: 'ItemsChanged', signature: 'a{sa{sv}}', sender: ':1.53', type: 4, flags: 1, body: [ [ [Array], [Array], [Array], [Array], [Array], [Array], [Array], [Array], [Array], [Array], [Array], [Array], [Array] ] ] }
Jun 02 10:59:04 2025-06-02T08:59:04.617Z node-red-contrib-victron:victron-client [MESSAGE HANDLER] com.victronenergy.vebus/276:/Dc/0/Power | { "changed": true, "path": "/Dc/0/Power", "value": 1181, "text": "1181W", "senderName": "com.victronenergy.vebus", "deviceInstance": 276 }

**Flow**
If applicable, add a flow to help explain your problem.

**Hardware (please complete the following information):**
 - Hardware: [e.g. Cerbo GX, Ekrano, Raspberry Pi]

**Software (please complete the following information):**
 - node-red-contrib-victron version:
 - Venus OS version:
 

**Additional context**
Add any other context about the problem here.

keeskode, 2025-06-08:
@dirkjanfaber thanks, i tested it but the debugging is still very verbose and floods logging. Seems nothing has changed. I like to re open the issue, but GitHub wont allows me to, as permission denied.

dirkjanfaber, 2025-06-08:
Assuming that logging is still about the same, that would come from here:
https://github.com/victronenergy/node-red-contrib-victron/blob/42228f275f7c75585617dea4860a8ab554ebeb22/src/services/dbus-listener.js#L292

Which is a clear `debug()` statement, which should only show if the `DEBUG` environment variable is set to either `*` or `node-red-contrib-victron` on startup of Node-RED. I suspect you have that somewhere in your startup code of Node-RED. 

How are you starting Node-RED on the Pi?

keeskode, 2025-06-09:
Hi Dirk Jan, I am using Node-Red as a plugin from SignalK, so that I have the embedded SignalK nodes. I did not changed anything in that setup that I can think has changed any DEBUG environment variable.

dirkjanfaber, 2025-06-10:
I am not that familiar with SignalK and the embedded Node-RED implementation there. Tagging @sbender9  in the hopes that he can provide some extra insight. As far as I can see, the code uses the correct way of setting the debugging to only `node-red-contrib-victron:dbus`. If that matches the filter set by SignalK starting Node-RED, that is beyond the scope of my control.

SignalK should use specific DEBUG patterns like `DEBUG=signalk:*` instead of `DEBUG=*`
Or explicitly exclude node-red-conrib-victron: `DEBUG=*,-victron:*,-node-red-contrib-victron*`

I cannot rule out that I am still overlooking something. Based on the info you provided, I cannot think of another cause than that SignalK somehow has the `DEBUG` variable set wrongly. 

As a plan B, you could comment out all of the debug calls (which you should then in the `node-red-contrib-victron` directory on your own install):
```
find src/ -name "*.js" -type f -exec sed -i 's/debug(/\/\/ debug(/g' {} \;
```