victronenergy/node-red-contrib-victron #246: [BUG]Virtual device: Dbus connection closed on 3.62

JeremyBloxham, 2025-06-27:
Creating a Virtual device on a Cerbo GX running 3.62 with node-red embedded in SignalK gives Dbus connection closed. With the standalone node-red it works.

sbender9, 2025-06-28:
I am having the same issue since updating to the latest palette. I'll do some debugging today...

sbender9, 2025-06-28:
@JeremyBloxham yours may be failing because `/data/home/nodered` does not exist.

Try logging on to the Cerbo and running `mkdir /data/home/nodered`

sbender9, 2025-06-28:
and this probably fixes it https://github.com/victronenergy/node-red-contrib-victron/commit/2b9dd5d3a58f81bd52968e3a02a9ca361978e15f

JeremyBloxham, 2025-06-28:
On the Cerbo, /data/home/nodered already exists.

Not immediately sure how to get the above mentioned fix onto my Cerbo, but I’ll dig into it.

Many thanks.

JeremyBloxham, 2025-06-28:
Yes, a Cerbo GX running 6.52. I've tried it on my Cerbo GX Mk 2 and on a Cerbo GX Mk 1 which I keep as a spare. On the Mk 1, I did a full re-install of the OS as a test and got the same issue.

This occurs with the embedded node-red installed from within SignalK.

If I run the node-red from the large image outside of SignalK it works fine (but then, of course, I can't use the SignalK nodes).

sbender9, 2025-06-28:
Can you check the Server Log after Signal K starts up? There should be some errors there.

(Might need to restart SK)

sbender9, 2025-06-28:
OK, so the real fix is https://github.com/victronenergy/node-red-contrib-victron/commit/2b9dd5d3a58f81bd52968e3a02a9ca361978e15f

Once that's released, you should be good.

In the mean time, setting this should work.

PERSISTED_STATE_LOCATION=/data/conf/signalk/red/Victorian

I think you can set this in node-red, under Settings, Environment

JeremyBloxham, 2025-06-28:
Fantastic -- that fixed it.

Many thanks indeed.

vandijks, 2025-06-29:
I have a similar situation:
- CerboGX with large firmware (3.62)
- NodeRed via SignalK (instead of Victron's implentation via Large FW).
- Virtual Motor Device for use in new Boat Display in FW 3.60
- "DBus connection closed" message on Virtual Motor Drive node.

I did follow the steps to add PERSISTED_STATE_LOCATION to Settings \ Environment. Rebooted, but no luck.

Error Log:

`Jun 29 10:20:25 === UNHANDLED REJECTION (PREVENTING CRASH) ===
Jun 29 10:20:26 Promise: Promise { <rejected> Error: EACCES: permission denied, mkdir '/data/home/nodered/.victron' at Object.mkdirSync (node:fs:1371:26) at hasPersistedState (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/persist.js:29:8) at proceed (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:838:13) at instantiateDbus (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1029:7) at new VictronVirtualNode (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1032:5) at Object.createNode (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/util.js:190:27) at Flow.start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/Flow.js:261:54) at async start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/index.js:403:17) { errno: -13, code: 'EACCES', syscall: 'mkdir', path: '/data/home/nodered/.victron' } }
Jun 29 10:20:26 Reason: Error: EACCES: permission denied, mkdir '/data/home/nodered/.victron' at Object.mkdirSync (node:fs:1371:26) at hasPersistedState (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/persist.js:29:8) at proceed (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:838:13) at instantiateDbus (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1029:7) at new VictronVirtualNode (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1032:5) at Object.createNode (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/util.js:190:27) at Flow.start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/Flow.js:261:54) at async start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/index.js:403:17) { errno: -13, code: 'EACCES', syscall: 'mkdir', path: '/data/home/nodered/.victron' }
Jun 29 10:20:26 Reason type: object
Jun 29 10:20:26 Is array: false
Jun 29 10:20:26 Stack trace:
Jun 29 10:20:26 Trace at process.<anonymous> (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:17:11) at process.emit (node:events:530:35) at emitUnhandledRejection (node:internal/process/promises:250:13) at throwUnhandledRejectionsMode (node:internal/process/promises:385:19) at processPromiseRejections (node:internal/process/promises:470:17) at process.processTicksAndRejections (node:internal/process/task_queues:96:32)
Jun 29 10:20:26 === END DEBUG ===
Jun 29 10:20:26 Unhandled Rejection at: Promise { <rejected> Error: EACCES: permission denied, mkdir '/data/home/nodered/.victron' at Object.mkdirSync (node:fs:1371:26) at hasPersistedState (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/persist.js:29:8) at proceed (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:838:13) at instantiateDbus (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1029:7) at new VictronVirtualNode (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1032:5) at Object.createNode (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/util.js:190:27) at Flow.start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/Flow.js:261:54) at async start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/index.js:403:17) { errno: -13, code: 'EACCES', syscall: 'mkdir', path: '/data/home/nodered/.victron' } } reason: Error: EACCES: permission denied, mkdir '/data/home/nodered/.victron' at Object.mkdirSync (node:fs:1371:26) at hasPersistedState (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/persist.js:29:8) at proceed (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:838:13) at instantiateDbus (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1029:7) at new VictronVirtualNode (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1032:5) at Object.createNode (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/util.js:190:27) at Flow.start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/Flow.js:261:54) at async start (/data/conf/signalk/node_modules/@node-red/runtime/lib/flows/index.js:403:17) { errno: -13, code: 'EACCES', syscall: 'mkdir', path: '/data/home/nodered/.victron' }
Jun 29 10:20:26 === UNHANDLED REJECTION (PREVENTING CRASH) ===
Jun 29 10:20:26 Promise: Promise { <rejected> Error: EACCES: permission denied, mkdir '/data/home/nodered/.victron' at Object.mkdirSync (node:fs:1371:26) at hasPersistedState (/data/conf/signalk/red/node_modules/@victronenergy/node-red-contr [... cut ...] e_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:1029:7) at /data/conf/signalk/red/node_modules/@victronenergy/node-red-contrib-victron/src/nodes/victron-virtual.js:396:11 { errno: -13, code: 'EACCES', syscall: 'mkdir', path: '/data/home/nodered/.victron' }
Jun 29 10:20:33 GET /plugins/signalk-node-red/redAdmin/red/images/typedInput/az.svg 200 49.861 ms - 846
Jun 29 10:20:57 POST /plugins/signalk-node-red/redAdmin/settings/user 204 30.437 ms - -`

Did I miss a step?

nmbath, 2025-07-02:
@vandijks Looks like it, see the first line, it can't create the required directory. This means that the setting has not been made correctly, or is pointing at the wrong location.

vandijks, 2025-07-04:
I can confirm that the latest version now works for me with the SignalK version of Node-Red. Thank you @dirkjanfaber!