victronenergy/node-red-contrib-victron #253: [regression?] issues with the GX built-in relays

mpvader, 2025-07-20:
Three things, welcome to split them

- [ ] Look into below bug (seems urgent!)
- [ ] While at it (?) Make below two nodes look similar.
- [ ] While at it, use the terminology GX built-in relay. And not venus or anything else like in below screenshots. See also our [manual of style](https://www.victronenergy.com/live/drafts:victron_manual_of_style?do=#list_of_words_that_are_often_written_incorrectly).

1. the regression, seems to be per v3.70~13 or 15, perhaps due to changes we made related to relay handling in Venus OS recently.

Its reported here: https://community.victronenergy.com/t/v3-70-15-node-red-relay-control-node-issue/39981, and I have it myself as well.

2. the top one of these two nodes seems to miss functionality compared to the bottom one. I can't give it a nice name:

[image]

3. two mentions of venus-device in services.json, needs to be GX device.

there is perhaps more, maybe grepping or searching?

[image]

mpvader, 2025-07-20:
This is one of the perhaps related changes to com.victronenergy.system.

The plan here was to make the backwards compatible wrt dbus API. So that Modbus-TCP and MQTT usage doesn’t break because of this

warwickchapman, 2025-07-20:
There are also some nodes where parameters are listed inside “Venus settings” category and others where they are in “Venus system”.

Ie. I think you open the node, choose the device, then choose either “Venus settings” or “Venus system” to expose the actual parameters. 

Eg. Active SoC is hidden on its own under one of those categories. 

I think that needs to be cleaned up. “Venus” should be dropped.

“GX device” should be fine for any settings limited to on the device as compared with, for example, settings on a “VE.Bus device” that may also be available on that node.

Chris927, 2025-07-20:
@mpvader Please split into separate issues, naming consistency and relay nodes regression are unrelated.